drndos/openspoolman #1: mqtt wont connect

prey87, 2024-12-13:
```
mqtt_bambulab.py", line 107, in async_subscribe
    MQTT_CLIENT.tls_set_context(ssl_ctx)
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'dict' object has no attribute 'tls_set_context'
```

It works when MQTT_CLIENT is not declared as a global variable.

```
def async_subscribe():
  MQTT_CLIENT = mqtt.Client()
```

First system macOS 15.1.1 Python 3.10.10
Second system Linux Debian 12.6 6.1.0-22-amd64 Python 3.11.2

prey87, 2024-12-13:
moving 
```
MQTT_CLIENT = {}  # Global variable storing MQTT Client
```
up to line 12 in mqtt_bambulab.py works

drndos, 2024-12-13:
Thank you, it worked somehow on windows so I didn't notice right away.