drndos/openspoolman #2: it is possible to run it on a container?

platini76, 2024-12-13:
the request is to have a simple installation for a container

prey87, 2024-12-13:
it's listed as a todo

drndos, 2024-12-13:
Yes, I have added a container and a compose file, helm file to come, and I will also check CI/CD so you can use docker image from github

drndos, 2024-12-13:
docker image is now available as a github package https://github.com/users/drndos/packages/container/package/openspoolman

platini76, 2024-12-14:
I am in a proxmox ct and vm environment.. 
how  to intall it please?

drndos, 2024-12-14:
It looks like it might be best to create a VM and install a docker/podman/other CR https://kubernetes.io/docs/setup/production-environment/container-runtimes/ there or if you need it for other stuff create a kubernetes cluster. 

If you don't want the kubernetes cluster you can start the CR on system startup `sudo systemctl enable docker` and add `restart: always` after 
```
services:
  openspoolman:
```
in compose.yaml, then simply run `docker-compose up -d` and it should run even if you restart the VM or CR environment until you run `docker-compose down`

If you fancy a kubernetes cluster the best way to start is IMHO microk8s. To run microk8s I have a quick how-to in Slovak language here: https://essential-data.sk/home-assistant-v-high-availability/ (if you use google translate, some of the code may be messed up, so check back to original page when copying code)

tretabyte, 2024-12-14:
there's already scripts to deploy a LXC container of Spoolman into proxmox easily, why not build a package, similar for example to what exists for Prowlarr or Sonarr and have it deployed into the Spoolman container.
I should save everyone a lot of headache having both things in the same place.

drndos, 2024-12-14:
I couldn't find the scripts you mentioned coudl you please link them? Unfortunately I don't have the proxmox environment in my homelab. Feel free to open PR :) Maybe then I could test it in my work dev environment where I have proxmox. Thanks

platini76, 2024-12-14:
here are many scrpts to make a ct or a vm with  a simple command
https://community-scripts.github.io/ProxmoxVE/

tretabyte, 2024-12-15:
> I couldn't find the scripts you mentioned could you please link them? Unfortunately I don't have the proxmox environment in my homelab. Feel free to open PR :) Maybe then I could test it in my work dev environment where I have proxmox. Thanks
> 
> Edit: you can use my helm chart of spoolman oci://ghcr.io/drndos/spoolman or edit the compose file to contain both spoolman and openspoolman together to make it easy and be together

https://raw.githubusercontent.com/community-scripts/ProxmoxVE/main/install/spoolman-install.sh

from https://community-scripts.github.io/ProxmoxVE/scripts?id=spoolman