drndos/openspoolman #3: [IDEA] Add QR Codes as an alternative to NFC

prey87, 2024-12-19:
Add the possibility to use QR codes as an alternative to NFC tags.

- Assign a random tag (e.g. XYZ123456) to each spool.
- Generate a URL like https://openspoolman.xyz/spool_info?tag_id=XYZ123456 and embed it into a QR code.

Advantages:

- No need to purchase NFC tags.
- QR codes can be scanned with any smartphone camera, making the setup more flexible.

Downside:
- If the base URL changes, all existing QR codes become invalid and need to be reprinted.

drndos, 2024-12-22:
Thank you for the suggestion! I am currently implementing the possibility to "fill the empty slot" from the UI, so you can search and find the filament from spoolman that you want to place to the slot without NFC tag.
I will look also into the QR codes and some sort of smart way to handle URL changes for QR codes and NFC tags.
I was also thinking about making a mobile app for people who don't have the environment to run the server.

d-nx, 2025-02-23:
Would it make sense using the brand's own barcode. I've noticed a lot of spools have their barcode handy on the spool itself and those should be pretty unique to that parcitular material/color.

Xento, 2025-02-26:
I added buttons to assign spools from a list.
View my fork.
I although created a pullrequest https://github.com/drndos/openspoolman/pull/20

Xento, 2025-03-26:
Checkout the latest version, there is no need for qr codes.
Just pick the tray and choose the spool or vice versa.

closed 2025-05-09 by Xento