drndos/openspoolman #10: docker container fails to start

briodan, 2025-01-20:
trying to get the docker container spun up but its not starting see below for error logs. the key one I think is this line as it seems to no include the Spoolman URL in a request

```
requests.exceptions.MissingSchema: Invalid URL 'None/api/v1/spool': No scheme supplied. Perhaps you meant https://None/api/v1/spool?
```

this is the docker config
```
version: '3.8'
services:
  openspoolman:
    image: ghcr.io/drndos/openspoolman:0.1.3
    container_name: openspoolman  
    ports:
      - "8080:8000"
```

env variables are set and the Spoolman base url is accessible

[image]

Full Log attached

[_openspoolman_logs.txt](https://github.com/user-attachments/files/18481272/_openspoolman_logs.txt)

logikal, 2025-01-22:
Looks like portainer? you need to actually get the env vars from portainer to your compose, like so:

```yaml
...
    environment:
      OPENSPOOLMAN_BASE_URL: ${OPENSPOOLMAN_BASE_URL}
      PRINTER_ACCESS_CODE: ${PRINTER_ACCESS_CODE}
      PRINTER_ID: ${PRINTER_ID}
      PRINTER_IP: ${PRINTER_IP}
      SPOOLMAN_BASE_URL: ${SPOOLMAN_BASE_URL}
      AUTO_SPEND: ${AUTO_SPEND}
```

briodan, 2025-01-23:
was using the provided env file to load the variables and that's how portainer does it. anyways not sure what changed, restarted the container and it started working.

separate question probably stupid, can't get it to write the nfc tag, when i got to the start writing NFC tag screen and scan with my android phone I get a pop-up from a separate apps with no record found but the tag does not get written, i'm sure its some kind of android setting application pairing but I have no idea what I'm doing lol.

closed 2025-01-24 by briodan