drndos/openspoolman #12: Error After Changing SPOOL in AMS

pad81, 2025-01-27:
When I insert a new spool into the AMS (full AMS, not lite) and use my NFC card assigned to a specific spool, I encounter the error: Tray color missing. Consequently, the spool is not automatically assigned in the AMS. To resolve this, I have to open the slicer or use the touchpad on the X1C to manually assign the color. This issue renders the NFC functionality ineffective.

If you need anything else, please let me know. I would love when everything is working good. 

It appears that the automatic spent filament is not consistently being added to Spoolman, with a success rate of only 1 out of 10 times.

buffdriver, 2025-01-27:
I believe that this is related to [Issue #9](https://github.com/drndos/openspoolman/issues/9)

pad81, 2025-01-28:
my bad. haven't seen it...sorry

buffdriver, 2025-01-28:
No worries, just highlighting for @drndos

Xento, 2025-04-15:
Is this stil an issue with the latest version?