drndos/openspoolman #13: Error 'id'

d-nx, 2025-02-07:
Deployed openspoolman in a container, the logs show a succesful connection to both printer (P1S) and Spoolman (shows spool & filament data), but the URL shows the following:

[image]

Printer firmware is: 01.06.01.02

Any help is greatly appreciated

`Connected with result code 5
Sent {'info': {'sequence_id': '0', 'command': 'get_version'}} to topic device/####/request
Sent {'pushing': {'sequence_id': '0', 'command': 'pushall'}} to topic device/####/request
Connected with result code 5
Sent {'info': {'sequence_id': '0', 'command': 'get_version'}} to topic device/####/request
Sent {'pushing': {'sequence_id': '0', 'command': 'pushall'}} to topic device/####/request
Connected with result code 5
Sent {'info': {'sequence_id': '0', 'command': 'get_version'}} to topic device/####/request
Sent {'pushing': {'sequence_id': '0', 'command': 'pushall'}} to topic device/####/request
Connected with result code 5
Sent {'info': {'sequence_id': '0', 'command': 'get_version'}} to topic device/####/request
Sent {'pushing': {'sequence_id': '0', 'command': 'pushall'}} to topic device/####/request
Connected with result code 5
Sent {'info': {'sequence_id': '0', 'command': 'get_version'}} to topic device/####/request
Sent {'pushing': {'sequence_id': '0', 'command': 'pushall'}} to topic device/####/request
200
Traceback (most recent call last):
  File "/home/app/app.py", line 138, in home
    augmentTrayDataWithSpoolMan(spool_list, vt_tray_data, trayUid(EXTERNAL_SPOOL_AMS_ID, vt_tray_data["id"]))
                                                                                         ~~~~~~~~~~~~^^^^^^
KeyError: 'id'`

weakfl, 2025-02-08:
You're firmware is quite old, the mqtt format has probably changed. Get something like [MQTTX](https://mqttx.app/) and check what the printer is sending.

There should be a `print` -> `vt_tray` property with external spool data. At least that's what openspoolman expects.

d-nx, 2025-02-08:
I just realized my printer code was wrong. Closing this. Thanks!

closed 2025-02-08 by d-nx