drndos/openspoolman #14: Provide arm64 docker image and make server port configurable

weakfl, 2025-02-09:
Would be nice if you could provide an `arm64` docker image.

Furthermore it would be nice if the server port would be configurable. `8000` is the same port `spoolman` uses and makes it cumbersome to configure nginx-proxy.

weakfl, 2025-02-11:
Tagging the `latest` image would be great too, to facilitate updating and enable watchtower usage.

Xento, 2025-04-19:
The latest tag is now implemented with https://github.com/drndos/openspoolman/pull/30

For the arm64 image I have to look how to do it.

weakfl, 2025-04-20:
Thanks again for your contributions to this great project @Xento!

Providing an arm64 image will be nice, but unfortunately of limited use without addressing #17 as well (imho). I will still have to build my own images to support the filament profiles I need.