drndos/openspoolman #17: Support custom filament profiles and improve profile handling

weakfl, 2025-02-16:
### Outline

Support custom filament profiles and improve support of system filament profiles.

### Details 

I have noticed that some of the system filament profiles are already handled in [filament.py](https://github.com/drndos/openspoolman/blob/f13e08e6c3c8c50f99a0137f876f36c4498d21ca/filament.py). Considering the vast amount of system profiles and profile variations (see [OrcaSlicer](https://github.com/SoftFever/OrcaSlicer/tree/main/resources/profiles/BBL/filament)), this looks like a maintenance nightmare imho.

I would like to propose to add another extra field to Spoolman, maybe something like `bbl_filament_profile_id`. If present use this field to set the `tray_info_idx` property to assign the filament profile.

This would not only allow to use system presets without having to add them in [filament.py](https://github.com/drndos/openspoolman/blob/f13e08e6c3c8c50f99a0137f876f36c4498d21ca/filament.py), but would also allow to use custom filament profiles.

The ids of custom profiles can easily be found in the slicer configs in `user/<id>/filament/base/<profile>.json` or via mqtt.

David8288, 2025-02-18:
Hello, 
Would it be possible to insert the profiles as in the xspool project?

Best regards
David 

https://xperiments.in/xspool/docs/sync

Xento, 2025-04-20:
I think with https://github.com/drndos/openspoolman/releases/tag/v0.1.8 it should work pretty well.
I found the project https://github.com/bofh69/spoolman2slicer but it does not support Bumbu Studio yet.

weakfl, 2025-04-21:
> I think with [v0.1.8 (release)](https://github.com/drndos/openspoolman/releases/tag/v0.1.8) it should work pretty well.

Works like a charm, thanks.

> I found the project [bofh69/spoolman2slicer](https://github.com/bofh69/spoolman2slicer) but it does not support Bumbu Studio yet.

Interesting, thanks. In my case I already have custom profiles for various filaments and I can select them just fine in Orca Slicer and Bambu Studio:

[image]

closed 2025-04-21 by weakfl