drndos/openspoolman #19: External spool erro

briodan, 2025-02-23:
Running into the error below, there is nothing assigned to the external spool slot

[image]

i get this or this

[image]

depending on how many times I reload?

There currently is nothing assigned to the external spool slot

```
Traceback (most recent call last):

  File "/home/app/app.py", line 138, in home

    augmentTrayDataWithSpoolMan(spool_list, vt_tray_data, trayUid(EXTERNAL_SPOOL_AMS_ID, vt_tray_data["id"]))

                                                                                         ~~~~~~~~~~~~^^^^^^

KeyError: 'id'
```

I tried assigning a spool to external slot in Spoolman and on the printer but nothing seems to have changed.

d-nx, 2025-02-23:
I ran into this issue and after some back and forth managed to get it to get tagged correctly.

Try tagging your the spool in spoolman (Active Tray field) using `YOURPRINTERSSERIAL_255_254` and see if that kicks things into gear.

briodan, 2025-02-23:
Yeah already did that but unfortunately has not recovered.

Xento, 2025-04-13:
Could you try the actual version if the issue persists?

briodan, 2025-04-15:
Updated the image a while back but forgot yo come back to this. Don’t seem to get this issue anymore with or without a spool set in the external slot.

closed 2025-04-15 by briodan