drndos/openspoolman #21: Spool tag not detected on backend

arshling, 2025-02-27:
When interacting directly with the openspoolman interface, I am correctly able to assign tags to my spools and it posts a "Success!" notification after scanning and assigning a spool. Spoolman also reflects the tag updates in the custom fields as well as in the active tray assignment with PRINTERSERIAL#_0_2

[image]

However, I'm getting continuous errors every couple of seconds that there is no tag assigned on the system backend that I've yet to be able to clear:

AMS [A] (hum: 3, temp: 21.2ºC)
    - [A2]  1F5C69FF (-01%) [[ 00000000000000000000000000000000 ]]
      - Not found. Update spool tag!

I've tried restarting the server/printer, double checked my access codes are correct, and tried resetting the AMS information after disconnecting down to one unit (I use two). Even though it looks correct from the webpage and reports 749g of filament left, it appears unable to track or update when printing because of the mismatch between what I'm seeing and what the server is reporting.

Server is running in venv on a Linux Mint miniPC, printer is a Bambu X1C in Lan Mode.

Happy to provide any other info needed.

Xento, 2025-02-28:
I think this is normal as you don't use a bambulab spool.
So the internal rfid reader from the ams can't read the tag.

> openspoolman_1  | AMS [A] (hum: 5, temp: 0.0ºC)
> openspoolman_1  |     - [A0]  000000FF (100%) [[ 00000000000000000000000000000000 ]]
> openspoolman_1  |       - Not found. Update spool tag!
> openspoolman_1  |     - [A1]  FFFAFAFF (100%) [[ 00000000000000000000000000000000 ]]
> openspoolman_1  |       - Not found. Update spool tag!
> openspoolman_1  |     - [A2]  0DFF00FF (100%) [[ 00000000000000000000000000000000 ]]
> openspoolman_1  |       - Not found. Update spool tag!
> openspoolman_1  |     - [A3]  C800FFFF (100%) [[ 00000000000000000000000000000000 ]]
> openspoolman_1  |       - Not found. Update spool tag!

arshling, 2025-03-01:
Ah, I see. I wondered if it might be the case that it was referring to the official uuid instead of the custom assigned tag, but I was a bit thrown off by the "Update spool tag!" message. Thank you for clarifying.

closed 2025-03-01 by arshling