drndos/openspoolman #25: Upgrading from version 0.1.4 to 0.1.5 cpu spikes to 100% constant usage

noismaster, 2025-04-07:
Upgrading from version 0.1.4 to 0.1.5 cpu spikes to 100% constant usage.

100% activity start when app connects to printer. If I change printers ip in config. It will not use 100% cpu.

Changing AUTO_SPEND to False doesn't change cpu usage.

[image]

noismaster, 2025-04-07:
I forgot to mention I'm running it in a docker container.

I'm not to familiar with python but I'll try to run it with PyCharm and see if I can find the reason. Seems to me that some loop is not sleeping and constantly processing.

weakfl, 2025-04-09:
I can confirm the issue:

```
CONTAINER ID   NAME                                   CPU %     MEM USAGE / LIMIT   MEM %     NET I/O           BLOCK I/O        PIDS
83d259013f1a   openspoolman                           100.34%   0B / 0B             0.00%     1.41MB / 66.7kB   1.63MB / 0B
```

Tagging @Xento as he implemented most of the recent changes. Could you maybe take a look @Xento?

Xento, 2025-04-09:
Thanks for the report.
I didn't noticed it.
It should be fixed by https://github.com/drndos/openspoolman/pull/26

drndos, 2025-04-09:
Merged in https://github.com/drndos/openspoolman/pull/26

closed 2025-04-09 by drndos