drndos/openspoolman #27: Adding more than one printer?

Amulqueen, 2025-04-09:
I have this as a docker compose file:
                                                                                                                     
version: "3"
services:
  openspoolman:
    image: ghcr.io/drndos/openspoolman:0.1.6
    container_name: openspoolman
    environment:
      OPENSPOOLMAN_BASE_URL: "http://1**.***.***.***:7914"
      PRINTER_ID: "01P00A*******"
      PRINTER_ACCESS_CODE: "*********"
      PRINTER_IP: "192.168.0.10"
      SPOOLMAN_BASE_URL: "http://1**.***.***.***:7912"
      AUTO_SPEND: "True"
    ports:
      - "7914:8000"
    volumes:
      - ./logs:/home/app/logs
      - ./data:/home/app/data
      - ./prints:/home/app/static/prints
    restart: unless-stopped

Is there a way to add another printeras i only have my p1s and would like my am1 to use this also.

Thanks

Xento, 2025-04-11:
At the moment nothing is prepared to handle a second printer.
Maybe it would be the easiest start to add a printername in the config.
So that you can see on which printer you are.
Maybe it would be possible to add links to the other container, so it is easier to jump between the two instances.

I'm on holiday the next two weeks, maybe I get something implemented.

weakfl, 2025-04-11:
Or you could use nginx-proxy and access the two printers via distinct paths, eh osm.foo.bar/p1s and osm.foo.bar/am1 (or distinct subdomains if you prefer that).

Xento, 2025-04-13:
Could you try my branch?
Now the default printer name and model is shown in the title.
A custom printername can be specified via env variable.
https://github.com/Xento/openspoolman

Amulqueen, 2025-04-16:
thanks im giving it a go now.