drndos/openspoolman #33: UndefinedError: 'None' has no attribute 'filament'

ginkel, 2025-04-22:
Hi there,

I just set up Spoolman + OpenSpoolMan. OpenSpoolMan is getting data from my AMS and then detects a mismatch it wants to fix:

[image]

When I click "Fix" on "Tray 1" an error page (Internal Server Error) is displayed and the log contains:

```
[2025-04-22 06:56:59,947] ERROR in app: Exception on /issue [GET]
Traceback (most recent call last):
  File "/home/app/venv/lib/python3.12/site-packages/flask/app.py", line 1511, in wsgi_app
    response = self.full_dispatch_request()
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/app/venv/lib/python3.12/site-packages/flask/app.py", line 919, in full_dispatch_request
    rv = self.handle_user_exception(e)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/app/venv/lib/python3.12/site-packages/flask/app.py", line 917, in full_dispatch_request
    rv = self.dispatch_request()
         ^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/app/venv/lib/python3.12/site-packages/flask/app.py", line 902, in dispatch_request
    return self.ensure_sync(self.view_functions[rule.endpoint])(**view_args)  # type: ignore[no-any-return]
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/app/app.py", line 58, in issue
    return render_template('issue.html', fix_ams=fix_ams, active_spool=active_spool)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/app/venv/lib/python3.12/site-packages/flask/templating.py", line 150, in render_template
    return _render(app, template, context)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/app/venv/lib/python3.12/site-packages/flask/templating.py", line 131, in _render
    rv = template.render(context)
         ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/app/venv/lib/python3.12/site-packages/jinja2/environment.py", line 1295, in render
    self.environment.handle_exception()
  File "/home/app/venv/lib/python3.12/site-packages/jinja2/environment.py", line 942, in handle_exception
    raise rewrite_traceback_stack(source=source)
  File "/home/app/templates/issue.html", line 1, in top-level template code
    {% extends 'base.html' %}
  File "/home/app/templates/base.html", line 108, in top-level template code
    {% block content %}{% endblock %}
    ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/app/templates/issue.html", line 5, in block 'content'
    {% with current_spool=active_spool %} {% include 'fragments/spool_details.html' %} {% endwith %}
    ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/app/templates/fragments/spool_details.html", line 10, in top-level template code
    style="background-color: #{{ current_spool.filament.color_hex }}; width: 40px; height: 40px; border-radius: 5px;">
    ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/app/venv/lib/python3.12/site-packages/jinja2/environment.py", line 490, in getattr
    return getattr(obj, attribute)
           ^^^^^^^^^^^^^^^^^^^^^^^
jinja2.exceptions.UndefinedError: 'None' has no attribute 'filament'
```

This is happening for all trays.

Any ideas?

Xento, 2025-04-22:
Just fill the tray with the desired spool.
The fix button is not working at the moment and needs some rework.
Maybe the fill button will be replaced by the fix button, when there is a mismatch or the fix button will reapply the chosen spool to the tray.

ginkel, 2025-04-22:
Works like a charm, thanks!