drndos/openspoolman #34: Local print from printer history does not detect AMS slot

ginkel, 2025-05-04:
Hi there,

when printing from my P1S's history ("cache") the AMS slot is not detected and thus the expended filament is not accounted to the correct (or any) spool:

[image]

Regards,
Thilo

Xento, 2025-05-04:
Could you upload the matching Mqtt log from the logs folder?

ginkel, 2025-05-04:
I hope the attached log is the right one. I can't find a perfect match to the timestamp in the UI, but considering all times seem to be in UTC, this should be the one.

[mqtt.log.gz](https://github.com/user-attachments/files/20029754/mqtt.log.gz)

Xento, 2025-05-05:
I think I fixed this issue but didn't had time to push it.
Did you use Bambu Studio 2.x?

Could you use the current version from github or the dev tag for the docker container?
https://ghcr.io/drndos/openspoolman