drndos/openspoolman #35: Is it possible to set filament_id for custom filaments to automatically see them in AMS?

piontec, 2025-05-05:
Hey!
First of all, thanks for the great project that saved spoolman for me after adding a bambu printer!
I have a question: last release adds option to set "filament_id" in spoolman. I was expecting this to allow me to automatically set my filament on "load" (assigning it in osm) in the orca slicer, but that doesn't seem to work. My custom filament definitions don't have "filament_id" property in their JSONs, only the base filaments do (the ones that orca automatically creates in "default" folder). Am I getting something wrong? Should this work? My custom filaments also get the "*.info" file, which looks like this
```
sync_info = update
user_id = 1719386693
setting_id = PFUSe95bdc93cd9a73
base_id = GFSNLS03_04
updated_time = 1745883094
```
so the "setting_id" looks like the ID created for the filament, but that doesn't work as well.
Also, sometimes orca creates my custom filaments in the "base/" filder, but most of them are 1 level up. Do you have any idea how this should be set up to make the "filament_id" field work?

Xento, 2025-05-05:
I didn't use orca slicer yet.
In spoolman you create the manufacturer, than the filament and then the spools.
So you could have one manufacturer which has multiple filaments which can be on multiple spools.
As the filamen_id is set on the filament you end up with multiple spools which have the same filament_id because they have the same filament on it.

For example I added all my spools to spoolman so that I know how many spools I have in my inventory left.
They all have the same filament_id as it is the same filament.

[image]

piontec, 2025-05-06:
That's the filament_id for spoolman, I mean using filament_id from orca, so that when I set my filament with OSM, it is correctly detected in the AMS in orca as well.

weakfl, 2025-05-06:
See https://github.com/drndos/openspoolman/issues/17#issue-2856032132

Easiest option is probably MQTT after assigning the filament profile on the AMS.

piontec, 2025-05-06:
I see this works for you! How did you do that? I checked `user/<id>/filament/base/<profile>.json ` and there are no "filament_id" properties there. How are you sniffing the MQTT traffic to check what's there?

piontec, 2025-05-06:
And, BTW, as I'm trying to understand how this works: do you know why/how orca sometimes puts my custome filament profiles into `filament/<profile>.json` and sometimes to `filament/base/<profile>.json`?

Xento, 2025-05-06:
I missunderstood you.
What you are trying to do should already work.
OpenSpoolMan puts the filament id via MQTT into the AMS.
The slicer than should read that out and select the correct filament profile.

I think I will install orca and test it on my site.

Maybe it depends how you create the profile, as I don't use Orca yet I don't know the workflow.
But maybe it its saved under filament/ when you take a base profile and use this as template.
When you create a profile from scratch there is no reference to a base profile so it will be saved under filament/base.

piontec, 2025-05-06:
OK, so here are a few cases I identified:

1. Using an already existing (in orca) pre-defined filament (I have flatpak install, so paths differ)
```
jq '.filament_id' ~/.config/OrcaSlicer/system/OrcaFilamentLibrary/filament/eSUN/eSUN\ PLA+\ @base.json 
"GFL03"
```
using this ID as assigned to my filament in spoolman works and correctly shows up in orca.

2. A filament created "from scratch" (I still don't get when one is "from scratch" and when an override in Orca)
```
➜  jq '.filament_id' ~/.var/app/io.github.softfever.OrcaSlicer/config/OrcaSlicer/user/1719386693/filament/base/Jayo\ PETG\ Black.json 
"null"
➜  cat ~/.var/app/io.github.softfever.OrcaSlicer/config/OrcaSlicer/user/1719386693/filament/base/Jayo\ PETG\ Black.info 
sync_info = 
user_id = 1719386693
setting_id = PFUS74c0ae04418a30
base_id = 
updated_time = 1746468077
```
No "filament_id" id .json. No base profile reference in .info. The `setting_id`, which seems like a profile, doesn't work

3. A new profile that is an override of existing one (I think this one gets created when you open an existing profile and then save it under a different name). This is my override for the existing eSUN filament from 1.
```
➜  cat ~/.var/app/io.github.softfever.OrcaSlicer/config/OrcaSlicer/user/1719386693/filament/SUNLU\ PLA+\ @BBL\ A1\ \(piontec\).json 
{
    "default_filament_colour": [
        "#57E389"
    ],
    "enable_pressure_advance": [
        "1"
    ],
    "filament_density": [
        "1.25"
    ],
    "filament_max_volumetric_speed": [
        "28"
    ],    
    "filament_settings_id": [
        "SUNLU PLA+ @BBL A1 (piontec)"
    ],
    "filament_type": [
        "PLA+"
    ],
    "from": "User",
    "inherits": "SUNLU PLA+ @BBL A1",
    "is_custom_defined": "0",
    "name": "SUNLU PLA+ @BBL A1 (piontec)",
    "pressure_advance": [
        "0.024"
    ],
    "supertack_plate_temp": [
        "50"
    ],
    "supertack_plate_temp_initial_layer": [
        "50"
    ],
    "version": "1.10.0.35"
}
➜  cat ~/.var/app/io.github.softfever.OrcaSlicer/config/OrcaSlicer/user/1719386693/filament/SUNLU\ PLA+\ @BBL\ A1\ \(piontec\).info 
sync_info = 
user_id = 1719386693
setting_id = PFUSe95bdc93cd9a73
base_id = GFSNLS03_04
updated_time = 1746468626
```
No filament_id and the `base_id` in .info is weird and not in sync with the ID in 1. :(

I hope this examples will make it easier for you to figure out how this should work.

weakfl, 2025-05-06:
>  How are you sniffing the MQTT traffic to check what's there?

You can use [MQTTX](https://mqttx.app/) to sniff the traffic (username is `bblp`, password is your printer access code).

Take a look at the `ams` property, specifically `ams` -> `tray` -> `tray_info_idx`

<details>

<summary>Expand for full JSON example</summary>

```json
"ams": [
    {
        "humidity": "5",
        "id": "0",
        "temp": "25.5",
        "tray": [
            {
                "bed_temp": "0",
                "bed_temp_type": "0",
                "cali_idx": -1,
                "cols": [
                    "3D3E3DFF"
                ],
                "ctype": 0,
                "drying_temp": "0",
                "drying_time": "0",
                "id": "0",
                "nozzle_temp_max": "280",
                "nozzle_temp_min": "220",
                "remain": -1,
                "tag_uid": "0000000000000000",
                "tray_color": "3D3E3DFF",
                "tray_diameter": "0.00",
                "tray_id_name": "",
                "tray_info_idx": "Pb5067e0",
                "tray_sub_brands": "",
                "tray_type": "PETG",
                "tray_uuid": "00000000000000000000000000000000",
                "tray_weight": "0",
                "xcam_info": "000000000000000000000000"
            },
            {
                "bed_temp": "0",
                "bed_temp_type": "0",
                "cali_idx": -1,
                "cols": [
                    "FFFFFFFF"
                ],
                "ctype": 0,
                "drying_temp": "0",
                "drying_time": "0",
                "id": "1",
                "nozzle_temp_max": "270",
                "nozzle_temp_min": "220",
                "remain": -1,
                "tag_uid": "0000000000000000",
                "tray_color": "FFFFFFFF",
                "tray_diameter": "0.00",
                "tray_id_name": "",
                "tray_info_idx": "GFG60",
                "tray_sub_brands": "",
                "tray_type": "PETG",
                "tray_uuid": "00000000000000000000000000000000",
                "tray_weight": "0",
                "xcam_info": "000000000000000000000000"
            },
            {
                "bed_temp": "0",
                "bed_temp_type": "0",
                "cali_idx": -1,
                "cols": [
                    "ADB1B2FF"
                ],
                "ctype": 0,
                "drying_temp": "65",
                "drying_time": "8",
                "id": "2",
                "nozzle_temp_max": "260",
                "nozzle_temp_min": "230",
                "remain": 14,
                "tag_uid": "6A4491F400000100",
                "tray_color": "ADB1B2FF",
                "tray_diameter": "1.75",
                "tray_id_name": "G02-D0",
                "tray_info_idx": "GFG02",
                "tray_sub_brands": "PETG HF",
                "tray_type": "PETG",
                "tray_uuid": "533A92B319DF4C73902F864592F65DE6",
                "tray_weight": "1000",
                "xcam_info": "803E803E8403E8030000803F"
            },
            {
                "bed_temp": "0",
                "bed_temp_type": "0",
                "cali_idx": -1,
                "cols": [
                    "00AE42FF"
                ],
                "ctype": 0,
                "drying_temp": "55",
                "drying_time": "8",
                "id": "3",
                "nozzle_temp_max": "230",
                "nozzle_temp_min": "190",
                "remain": 84,
                "tag_uid": "BAFDA73B00000100",
                "tray_color": "00AE42FF",
                "tray_diameter": "1.75",
                "tray_id_name": "A00-G6",
                "tray_info_idx": "GFA00",
                "tray_sub_brands": "PLA Basic",
                "tray_type": "PLA",
                "tray_uuid": "7B653B1C6E5440F7BCF2A66E0B07F5BE",
                "tray_weight": "1000",
                "xcam_info": "88138813E803E8039A99193F"
            }
        ]
    }
],
```
</details>

> And, BTW, as I'm trying to understand how this works: do you know why/how orca sometimes puts my custome filament profiles into filament/<profile>.json and sometimes to filament/base/<profile>.json?

The ones you want are the base ones, which are custom filaments and not just profiles. You create a custom filament for a manufacturer and assign a profile to it:

1. Click here:

[image]

2. Create a custom filament:

[image]

Then you'll find the id in the `filament_id` property in `filament/base/<profile>.json`.

And in Spoolman you assign the `filament_id` on the filament level, not on the spool level.

Hope that helps...

piontec, 2025-05-06:
Gee, so I was creating my filaments wrong... Still, when I create a new filament, it deosn't get the `filament_id` automatically, it seems... I'll try to add it manually. If you have the `filament_id` in the .json file already, I guess it's enough to add it in the spoolman, right? I don't even need anything from MQTT, I think.

Thanks a lot for this explanaition!

weakfl, 2025-05-06:
>  If you have the filament_id in the .json file already, I guess it's enough to add it in the spoolman, right? 

Yes, you assign the ID to the filament in Spoolman:

[image]

> Still, when I create a new filament, it deosn't get the filament_id automatically, it seems... I'll try to add it manually.

I think mine got an ID automatically, but it's been a while so I'm not sure tbh.

> I don't even need anything from MQTT, I think.

Not necessarily, it's just convenient if you want to get the ID of one of the existing profiles (eg. Polymaker). You can also check the [OrcaSlicer repository](https://github.com/SoftFever/OrcaSlicer/tree/main/resources/profiles/BBL/filament) for IDs.

> Thanks a lot for this explanaition!

Welcome!

closed 2025-05-08 by piontec