drndos/openspoolman #37: Error when fixing mismatching AMS Slot

simonszu, 2025-05-14:
This occurs on a new setup, so maybe this is a layer 8 error. However, i have created some spools in spoolman which resemble the spools currently in my AMS. Openspoolman detects these spools, and i can fill them to the respective slots in Openspoolman. None of these spools have NFC. 

I get a warning: 

'''
There is a mismatch between printer and SpoolMan
Fix issue by clicking "Fix this tray" on tray with !
'''

No matter on which slot i click, i get presented an Internal Server Error page. The log shows  this stacktrace: 

'''
[2025-05-14 09:59:07,537] ERROR in app: Exception on /issue [GET]
Traceback (most recent call last):
  File "/home/app/venv/lib/python3.12/site-packages/flask/app.py", line 1511, in wsgi_app
    response = self.full_dispatch_request()
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/app/venv/lib/python3.12/site-packages/flask/app.py", line 919, in full_dispatch_request
    rv = self.handle_user_exception(e)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/app/venv/lib/python3.12/site-packages/flask/app.py", line 917, in full_dispatch_request
    rv = self.dispatch_request()
         ^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/app/venv/lib/python3.12/site-packages/flask/app.py", line 902, in dispatch_request
    return self.ensure_sync(self.view_functions[rule.endpoint])(**view_args)  # type: ignore[no-any-return]
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/app/app.py", line 58, in issue
    return render_template('issue.html', fix_ams=fix_ams, active_spool=active_spool)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/app/venv/lib/python3.12/site-packages/flask/templating.py", line 150, in render_template
    return _render(app, template, context)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/app/venv/lib/python3.12/site-packages/flask/templating.py", line 131, in _render
    rv = template.render(context)
         ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/app/venv/lib/python3.12/site-packages/jinja2/environment.py", line 1295, in render
    self.environment.handle_exception()
  File "/home/app/venv/lib/python3.12/site-packages/jinja2/environment.py", line 942, in handle_exception
    raise rewrite_traceback_stack(source=source)
  File "/home/app/templates/issue.html", line 1, in top-level template code
    {% extends 'base.html' %}
  File "/home/app/templates/base.html", line 108, in top-level template code
    {% block content %}{% endblock %}
    ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/app/templates/issue.html", line 5, in block 'content'
    {% with current_spool=active_spool %} {% include 'fragments/spool_details.html' %} {% endwith %}
    ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/app/templates/fragments/spool_details.html", line 10, in top-level template code
    style="background-color: #{{ current_spool.filament.color_hex }}; width: 40px; height: 40px; border-radius: 5px;">
    ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/app/venv/lib/python3.12/site-packages/jinja2/environment.py", line 490, in getattr
    return getattr(obj, attribute)
           ^^^^^^^^^^^^^^^^^^^^^^^
jinja2.exceptions.UndefinedError: 'None' has no attribute 'filament'
'''

Xento, 2025-05-14:
The fix button isn't working and I think we can remove it or replace it with the fill buttons function.
What happens, when you reassign the correct spool?

Openspoolman checks if a spool is assigned and than checks for the filament type if it matches the spool.

Could you post a picture of openspoolman home view and than a picture of the spool settings in spoolman?

Xento, 2025-05-25:
Does the fill button fix your issue?

Xento, 2025-05-30:
Can you take screenshots from the active_tray attribute of your selected spools in spoolman?

simonszu, 2025-06-02:
I could, for sure, but where do i find this attribute? Is it the "Location" one?

Xento, 2025-06-02:
No you have to add it manually https://github.com/drndos/openspoolman?tab=readme-ov-file#how-to-setup