drndos/openspoolman #39: Invalid Host

korniza, 2025-05-25:
I try to run the service on a debian 12 (ip: 192.168.96.x) and I get the following error. 
I have a Bambu A1 printer (ip: 192.168.15.x). I double checked all the IP addresses and variables. Any clue? 

$ python3 wsgi.py
```
 * Serving Flask app 'app'
 * Debug mode: off
🔄 Trying to connect ...
⚠️ connection failed: Invalid host., new try in 15 seconds...
WARNING: This is a development server. Do not use it in a production deployment. Use a production WSGI server instead.
 * Running on all addresses (0.0.0.0)
 * Running on https://127.0.0.1:8443
 * Running on https://192.168.97.9:8443
Press CTRL+C to quit
🔄 Trying to connect ...
⚠️ connection failed: Invalid host., new try in 15 seconds...
```

Xento, 2025-05-25:
You have an unusual IP range. 
Are you sure that the server and the printer are in the same network or the routes are correctly configured?

korniza, 2025-05-25:
there are 2 networks that are routed between them.

Xento, 2025-05-26:
It seems that the container has its own ipadress. Does the container know the route to the other network?

korniza, 2025-05-29:
I have installed on Unraid. for first time I get the web UI up. Still I get 
"Error MQTT is disconnected. Is the printer online?"
My device is a Bambu A1 with latest firmware (01.04.0000) and with AMS-lite.
Still no clue. Hope someone to build a working container for Unraid.

closed 2025-08-17 by Xento

leisegang, 2025-08-30:
I have this error to. Spoolman is on 192.168.0.100 the printer has 192.168.0.39 and trying to get the open spoolman on 192.168.0.50 it gives the same error as here. And gives a error 500 on chrome