drndos/openspoolman #40: Unable to start container

platini76, 2025-06-07:
`WARN[0000] /opt/stacks/openspoolman/compose.yaml: the attribute `version` is obsolete, it will be ignored, please remove it to avoid potential confusion 
openspoolman-1  | [2025-06-08 00:38:15 +0200] [7] [ERROR] Exception in worker process
openspoolman-1  | Traceback (most recent call last):
openspoolman-1  |   File "/home/app/venv/lib/python3.12/site-packages/gunicorn/arbiter.py", line 608, in spawn_worker
openspoolman-1  |     worker.init_process()
openspoolman-1  |   File "/home/app/venv/lib/python3.12/site-packages/gunicorn/workers/gthread.py", line 94, in init_process
openspoolman-1  |     super().init_process()
openspoolman-1  |   File "/home/app/venv/lib/python3.12/site-packages/gunicorn/workers/base.py", line 135, in init_process
openspoolman-1  |     self.load_wsgi()
openspoolman-1  |   File "/home/app/venv/lib/python3.12/site-packages/gunicorn/workers/base.py", line 147, in load_wsgi
openspoolman-1  |     self.wsgi = self.app.wsgi()
openspoolman-1  |                 ^^^^^^^^^^^^^^^
openspoolman-1  |   File "/home/app/venv/lib/python3.12/site-packages/gunicorn/app/base.py", line 66, in wsgi
openspoolman-1  |     self.callable = self.load()
openspoolman-1  |                     ^^^^^^^^^^^
openspoolman-1  |   File "/home/app/venv/lib/python3.12/site-packages/gunicorn/app/wsgiapp.py", line 57, in load
openspoolman-1  |     return self.load_wsgiapp()
openspoolman-1  |            ^^^^^^^^^^^^^^^^^^^
openspoolman-1  |   File "/home/app/venv/lib/python3.12/site-packages/gunicorn/app/wsgiapp.py", line 47, in load_wsgiapp
openspoolman-1  |     return util.import_app(self.app_uri)
openspoolman-1  |            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
openspoolman-1  |   File "/home/app/venv/lib/python3.12/site-packages/gunicorn/util.py", line 370, in import_app
openspoolman-1  |     mod = importlib.import_module(module)
openspoolman-1  |           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
openspoolman-1  |   File "/usr/local/lib/python3.12/importlib/__init__.py", line 90, in import_module
openspoolman-1  |     return _bootstrap._gcd_import(name[level:], package, level)
openspoolman-1  |            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
openspoolman-1  |   File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
openspoolman-1  |   File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
openspoolman-1  |   File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
openspoolman-1  |   File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
openspoolman-1  |   File "<frozen importlib._bootstrap_external>", line 999, in exec_module
openspoolman-1  |   File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
openspoolman-1  |   File "/home/app/app.py", line 11, in <module>
openspoolman-1  |     from mqtt_bambulab import fetchSpools, getLastAMSConfig, publish, getMqttClient, setActiveTray, isMqttClientConnected, init_mqtt, getPrinterModel
openspoolman-1  |   File "/home/app/mqtt_bambulab.py", line 10, in <module>
openspoolman-1  |     from spoolman_service import spendFilaments, setActiveTray, fetchSpools
openspoolman-1  |   File "/home/app/spoolman_service.py", line 5, in <module>
openspoolman-1  |     from print_history import update_filament_spool
openspoolman-1  |   File "/home/app/print_history.py", line 150, in <module>
openspoolman-1  |     create_database()
openspoolman-1  |   File "/home/app/print_history.py", line 15, in create_database
openspoolman-1  |     conn = sqlite3.connect(db_config["db_path"])
openspoolman-1  |            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
openspoolman-1  | sqlite3.OperationalError: unable to open database file
openspoolman-1  | [2025-06-08 00:38:15 +0200] [7] [INFO] Worker exiting (pid: 7)
openspoolman-1  | [2025-06-08 00:38:15 +0200] [1] [ERROR] Worker (pid:7) exited with code 3
openspoolman-1  | [2025-06-08 00:38:15 +0200] [1] [ERROR] Shutting down: Master
openspoolman-1  | [2025-06-08 00:38:15 +0200] [1] [ERROR] Reason: Worker failed to boot.
openspoolman-1  | [2025-06-08 00:38:41 +0200] [1] [INFO] Starting gunicorn 23.0.0
openspoolman-1  | [2025-06-08 00:38:41 +0200] [1] [INFO] Listening at: http://0.0.0.0:8000 (1)
openspoolman-1  | [2025-06-08 00:38:41 +0200] [1] [INFO] Using worker: gthread
openspoolman-1  | [2025-06-08 00:38:41 +0200] [7] [INFO] Booting worker with pid: 7
openspoolman-1  | [2025-06-08 00:38:42 +0200] [7] [ERROR] Exception in worker process
openspoolman-1  | Traceback (most recent call last):
openspoolman-1  |   File "/home/app/venv/lib/python3.12/site-packages/gunicorn/arbiter.py", line 608, in spawn_worker
openspoolman-1  |     worker.init_process()
openspoolman-1  |   File "/home/app/venv/lib/python3.12/site-packages/gunicorn/workers/gthread.py", line 94, in init_process
openspoolman-1  |     super().init_process()
openspoolman-1  |   File "/home/app/venv/lib/python3.12/site-packages/gunicorn/workers/base.py", line 135, in init_process
openspoolman-1  |     self.load_wsgi()
openspoolman-1  |   File "/home/app/venv/lib/python3.12/site-packages/gunicorn/workers/base.py", line 147, in load_wsgi
openspoolman-1  |     self.wsgi = self.app.wsgi()
openspoolman-1  |                 ^^^^^^^^^^^^^^^
openspoolman-1  |   File "/home/app/venv/lib/python3.12/site-packages/gunicorn/app/base.py", line 66, in wsgi
openspoolman-1  |     self.callable = self.load()
openspoolman-1  |                     ^^^^^^^^^^^
openspoolman-1  |   File "/home/app/venv/lib/python3.12/site-packages/gunicorn/app/wsgiapp.py", line 57, in load
openspoolman-1  |     return self.load_wsgiapp()
openspoolman-1  |            ^^^^^^^^^^^^^^^^^^^
[... 327 lines cut ...]
openspoolman-1  |   File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
openspoolman-1  |   File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
openspoolman-1  |   File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
openspoolman-1  |   File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
openspoolman-1  |   File "<frozen importlib._bootstrap_external>", line 999, in exec_module
openspoolman-1  |   File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
openspoolman-1  |   File "/home/app/app.py", line 11, in <module>
openspoolman-1  |     from mqtt_bambulab import fetchSpools, getLastAMSConfig, publish, getMqttClient, setActiveTray, isMqttClientConnected, init_mqtt, getPrinterModel
openspoolman-1  |   File "/home/app/mqtt_bambulab.py", line 10, in <module>
openspoolman-1  |     from spoolman_service import spendFilaments, setActiveTray, fetchSpools
openspoolman-1  |   File "/home/app/spoolman_service.py", line 5, in <module>
openspoolman-1  |     from print_history import update_filament_spool
openspoolman-1  |   File "/home/app/print_history.py", line 150, in <module>
openspoolman-1  |     create_database()
openspoolman-1  |   File "/home/app/print_history.py", line 15, in create_database
openspoolman-1  |     conn = sqlite3.connect(db_config["db_path"])
openspoolman-1  |            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
openspoolman-1  | sqlite3.OperationalError: unable to open database file
openspoolman-1  | [2025-06-07 22:41:30 +0000] [7] [INFO] Worker exiting (pid: 7)
openspoolman-1  | [2025-06-07 22:41:30 +0000] [1] [ERROR] Worker (pid:7) exited with code 3
openspoolman-1  | [2025-06-07 22:41:30 +0000] [1] [ERROR] Shutting down: Master
openspoolman-1  | [2025-06-07 22:41:30 +0000] [1] [ERROR] Reason: Worker failed to boot.
openspoolman-1 exited with code 3`

please someone can help me?

platini76, 2025-06-09:
I cannot downgrade with H2D, the only option I have it to wait new version

Xento, 2025-06-15:
Are you sure you added the extra fields to spoolman?
I looked into the code and I'm pretty sure it should set the spools.
The only thing that doesn't works is setting the filament color and type.