drndos/openspoolman #42: Write tags from iOS?

Drudoo, 2025-06-25:
The readme says it is difficult to write tags from iOS, as that’s the only device I got, how would I do it? I’m used to making NFC tags for other things and normally use “NFC tools” or “TagWriter” but neither app contains a browser. 

Is it possible to get the info from openspoolman and manually type it into a writer app and then write the tag?

Drudoo, 2025-06-26:
So what you need to write to the tag is basically `http://base_url:port/spool_info?spool_id=x` where x is the spool ID in openspoolman?

weakfl, 2025-06-27:
I don't remember the exact URL tbh, and am afk atm. I looked it up in the source code at the time.

But if it works entering manually in Safari it will work just as well with the tag. And you can always overwrite it anyway.

Drudoo, 2025-06-27:
So what you have to write is `http://base_url:port/spool_info?tag_id=x` where x is the Extra Fields' tag from SpoolMan. it will be a uuid.