drndos/openspoolman #43: AMS picking wrong filament

Drudoo, 2025-06-27:
I have successfully write 4 uuids to NFC tags and stuck them to 4 rolls. 

When i insert a roll of filament, it loads but it doesn't pick the right color and type, it just loads when was last in that spot. It seems like OpenSpoolMan doesn't send the right info to the printer.

Drudoo, 2025-06-29:
Alright, I must have misunderstood the functionality. 

Where in the handy app do you scan it? I can’t see anywhere to do so.

weakfl, 2025-06-29:
> Where in the handy app do you scan it? I can’t see anywhere to do so.

Just hold the top of your iPhone near the tag. You'll be prompted if you want to open the URL.
Accept and it will open the page of the specific filament and you can assign it to an AMS slot.

Drudoo, 2025-06-29:
Okay, I see. But that only changes the filament type in OpenSpoolMan and not in the actual Bambu software. 

If nothing is changed in Bambu, what’s the point?

Xento, 2025-06-29:
It changes it in Bambustudio if you are not on the latest firmware. When you are on the latest firmware you have to enable lan only mode.
This is the new "security" system from bambulab. This is the limitation for all even openspool. 
But openspoolman tracks the used filament in spoolman.