drndos/openspoolman #44: only Tray 1 spool assigned

Helloookimi, 2025-07-11:
Thank you for creating such an excellent tool for calculating filament remaining weight. 
During my installation and debugging, I discovered an issue. All parameters, including the SpoolMan,Filament ID, were set correctly. However, when I place any spool in Tray 1, it is accurately recorded. But when placed in the other three trays, it fails to be tracked properly. After analyzing the 3MF file, I noticed that Bambu Studio consistently outputs filamentOrder': {1: 0}} regardless of tray location. Could this be the root cause?
I also tested this behavior with Orca, and observed that the 3MF file there contains filamentOrder values like 1,1,2,3, instead. Below is the raw code snippet.any ideas on how to resolve this inconsistency? Thank you again for your work.
bambu studio
Tray 1:
Downloading 3MF file from cloud...
3MF file downloaded and saved as /tmp/tmpkkccx5qh.3mf.
{'file': 'CNab52d631a7543a_68457350_4.3mf', 'plateID': '4', 'filaments': {1: {'id': 1, 'tray_info_idx': 'P7358617', 'type': 'PETG', 'color': '#FA75B5', 'used_g': '1.25', 'used_m': '0.41'}}, 'usage': {1: '1.25'}, 'image': '20250712015916.png', 'filamentOrder': {1: 0}}
200
Tray 2:
3MF file downloaded and saved as /tmp/tmpkhx__oje.3mf.
{'file': 'CN2ff0cd2e2764d0_68457537_7.3mf', 'plateID': '7', 'filaments': {1: {'id': 1, 'tray_info_idx': 'P7358617', 'type': 'PETG', 'color': '#E91F26', 'used_g': '1.59', 'used_m': '0.52'}}, 'usage': {1: '1.59'}, 'image': '20250712020135.png', 'filamentOrder': {1: 0}}
200
Tray 3:
Downloading 3MF file from cloud...
3MF file downloaded and saved as /tmp/tmpwx_5iz7c.3mf.
{'file': 'CNda58a05d72fe09_68457639_5.3mf', 'plateID': '5', 'filaments': {1: {'id': 1, 'tray_info_idx': 'P7358617', 'type': 'PETG', 'color': '#E4D2B6', 'used_g': '12.92', 'used_m': '4.23'}}, 'usage': {1: '12.92'}, 'image': '20250712020311.png', 'filamentOrder': {1: 0}}
Tray 4：
Downloading 3MF file from cloud...
3MF file downloaded and saved as /tmp/tmpzwuhkf1s.3mf.
{'file': 'CN21bde21ff94405_68457753_7.3mf', 'plateID': '7', 'filaments': {1: {'id': 1, 'tray_info_idx': 'P7358617', 'type': 'PETG', 'color': '#A07544', 'used_g': '1.59', 'used_m': '0.52'}}, 'usage': {1: '1.59'}, 'image': '20250712020437.png', 'filamentOrder': {1: 0}}

orca：
Tray 1:
Downloading 3MF file from cloud...
3MF file downloaded and saved as /tmp/tmpv1ef1my_.3mf.
{'file': 'CN2240d79a449208_68455968_4.3mf', 'plateID': '4', 'filaments': {1: {'id': 1, 'tray_info_idx': 'P7358617', 'type': 'PETG', 'color': '#FA75B5', 'used_g': '1.75', 'used_m': '0.57'}}, 'usage': {1: '1.75'}, 'image': '20250712014015.png', 'filamentOrder': {1: 0}}
Tray 2:
Downloading 3MF file from cloud...
3MF file downloaded and saved as /tmp/tmpoykui12c.3mf.
{'file': 'CN6ecbc335746351_68456769_6.3mf', 'plateID': '6', 'filaments': {1: {'id': 1, 'tray_info_idx': 'P7358617', 'type': 'PETG', 'color': '#E91F26', 'used_g': '1.26', 'used_m': '0.41'}}, 'usage': {1: '1.26'}, 'image': '20250712015125.png', 'filamentOrder': {1: 0}}
Tray 3:
Downloading 3MF file from cloud...
3MF file downloaded and saved as /tmp/tmp0afuwmm0.3mf.
{'file': 'CN5103c3d6cd0ffc_68455605_5.3mf', 'plateID': '5', 'filaments': {1: {'id': 1, 'tray_info_idx': 'P7358617', 'type': 'PETG', 'color': '#E4D2B6', 'used_g': '13.15', 'used_m': '4.30'}}, 'usage': {1: '13.15'}, 'image': '20250712013445.png', 'filamentOrder': {2: 0}}
Tray 4：
Downloading 3MF file from cloud...
3MF file downloaded and saved as /tmp/tmpwh_6q0by.3mf.
{'file': 'CNcb6ae9dcdcb5d3_68457031_7.3mf', 'plateID': '7', 'filaments': {1: {'id': 1, 'tray_info_idx': 'P7358617', 'type': 'PETG', 'color': '#A07544', 'used_g': '1.60', 'used_m': '0.52'}}, 'usage': {1: '1.60'}, 'image': '20250712015445.png', 'filamentOrder': {3: 0}}

Helloookimi, 2025-07-13:
when I use four spools at one time,  I find 3mf file is right
3MF file downloaded and saved as /tmp/tmp22cr57_k.3mf.
{'file': 'CNe71f7fd5ee8772_68835678_6.3mf', 'plateID': '6', 'filaments': {1: {'id': 1, 'tray_info_idx': 'P7358617', 'type': 'PETG', 'color': '#A07544', 'used_g': '6.95', 'used_m': '2.28'}, 2: {'id': 2, 'tray_info_idx': 'P7358617', 'type': 'PETG', 'color': '#121212', 'used_g': '2.98', 'used_m': '0.98'}, 3: {'id': 3, 'tray_info_idx': 'P7358617', 'type': 'PETG', 'color': '#0000FF', 'used_g': '2.56', 'used_m': '0.84'}, 4: {'id': 4, 'tray_info_idx': 'P7358617', 'type': 'PETG', 'color': '#FED034', 'used_g': '29.31', 'used_m': '9.59'}}, 'usage': {1: '6.95', 2: '2.98', 3: '2.56', 4: '29.31'}, 'image': '20250714005054.png', 'filamentOrder': {6: 0, 5: 1, 7: 2, 3: 3}}
however it still cannot be assigned in openspoolman

nightstorm2, 2025-07-26:
I have the same problem with BambuStudio and my P1S with my AMS. The first filament is correctly assigned to the printed model / filament in SpoolMan.

Xento, 2026-01-01:
Is this stil an issue with the latest version?